CentyLab/PocketPD #4: Where is the schematic?

Kabron287, 2024-11-27:
Where is the schematic?

CentyLab, 2025-03-25:
Hi, we will release the hardware files once the product is launched for a production run. Please stay toon.

LokiMetaSmith, 2025-05-08:
Yeah, I'm interested in integrating this into either stand alone or purpose built odd ball tools that a general purpose 90w power supply would support. Features I'm looking for are GPIO to turn this thing into a hat or similar, a way to control it over i2c or similar. 

for example, I think this would make a lovely electroplating pencil in CC mode, a felt fiber pen, and a few electrodes. Could electroplate copper/nickel/gold without needing a precision lab power supply, and it's flexible enough and low wattage enough to be portable.

Kabron287, 2025-05-08:
This project is about to die and they haven't posted the schematics yet

LokiMetaSmith, 2025-05-09:
That's a shame, it looks like I can design something similar pretty easily, I just was hoping to get something off the shelf before doing a fresh design. 

It also appears there are more options on Amazon than there use to be, just not open source😒

CentyLab, 2025-05-09:
Please find the hardware files at [PocketPD_HW](https://github.com/CentyLab/PocketPD_HW). 
Sorry for the wait.

closed 2025-05-09 by CentyLab