CentyLab/PocketPD #5: Wrongly display PPS mode and show voltage change

CentyLab, 2024-12-04:
Seen in version 0.9.5
This behavior can be replicate by:
+ Bootup in fix PDO
+ Going to Menu by long press Volt/Amp button
+ Rotate the encoder
+ Cancel voltage change by long press Volt/Amp button

=> Display PPS mode and allow voltage target/current limit to change.

IDKhowtousethissiteLol, 2025-04-18:
Has this been updated yet? I have purchsead a pocketpd and the profiles still dont select, and my firmware is 0.9.7. Also I noticed that the newest firmware isnt on the github.

CentyLab, 2025-04-19:
Hi @IDKhowtousethissiteLol ,
The issue is still exist in 0.9.7. We will work to have it fixed in 0.9.8 release. 
You can find 0.9.7 in the Development branch and not yet release as we still need to work on ReadMe.md