CentyLab/PocketPD #7: No Profiles Detected - Apple 70W USB-C Charger

matt6626, 2025-04-19:
Just went to test that my HW1.1 PocketPD works without yet having a PPS compatible charger.

My assumption is that the Apple 70W USB-C charger should probably correctly support at least some standard PDO profiles, but unable to get any output and “No profile detected” on bootup.

Haven’t tried digging through the source code at this stage to understand the expected operation.

CentyLab, 2025-04-19:
@matt6626 This is unexpected. We don't have the exact charger, but some lower wattage one for MacBook Air that we will be able to test sometime next week. Will keep you posted.

matt6626, 2025-04-19:
I have now tested a second unit, and the second unit works ok. So it’s presumably some hardware issue with the unit in question given they are both on the same fw (0.9.7) that the units came with.

Will open the unit to inspect at some point.

CentyLab, 2025-04-26:
The unit has been confirmed to have hardware issue. Thank you @matt6626

closed 2025-04-26 by CentyLab